Smoothieware/Smoothieboard2 #18: Where is schematic and pcb?

sjdaliyan, 2017-02-08:
I want to plan upgrade for v2.
How can i look at schematic and pcb?

arthurwolf, 2017-02-08:
v2 is not at the point where you can upgrade to it. We do not even have an actual prototype, and we do not have a firmware.

closed 2017-02-08 by arthurwolf

sjdaliyan, 2017-02-08:
Hey Arthur why don't you answer my emails?

arthurwolf, 2017-02-08:
I had some emergencies with customer's machines, and I am late on email, I have about 300 emails to answer at the moment. I'll get to yours eventually.

sjdaliyan, 2017-02-08:
ok.

I need an simple version of smoothie board for start.
Can i use newer versions of smoothie board on Sunbeam?
Sunbeam is simpler version of smoothieboard,right?
What about SmoothieBrainz?
I'm still have the same problem with my self made prototype.

arthurwolf, 2017-02-08:
I'm not sure I understand.
Sunbeam is a derivative of Smoothieboard, it's not really "simpler".
Same thing for "SmoothieBrainz".
Why not use a Smoothieboard ? You seem to require a lot of help, and it's easier to get help if you have a Smoothieboard ...

sjdaliyan, 2017-02-08:
Yes i need help,Mostly because of how i'm spending time on smoothie board.
As i told you i have a 1768 header board.